adafruit/circuitpython-org #1115: How to put warning about board?

dglaude, 2022-12-12:
Not all board offer the same user experience one might expect from a CircuitPython board.

So I was wondering if a proper way to indicate those potential problem or surprise could be defined. This idea came while playing with this board: #1114 

Here are some "glitch":
1) JST SH vs JST PH: some early board do not that the QT connector, but the big one
2) Some board have a JST PH connector, but do not use Adafruit pinout
3) 5V vs 3V: some board you have to select voltage or are 5V and you want to know
4) Some board with build in display 
5) Some board do not have full native USB and don't have mass storage (C3 and ESP32 board)
6) Some board do not have double click reset (nor boot button) to get in UF2 upgrade mode, but require special software, or connecting two pin, or ...
7) Some board have Feather format, but do not respect the usual M4 pin numbering

When this is an Adafruit board, the caveats can be in the learn guide, but for non-Adafruit board the download page could help identify those small or bit annoyance (and make the benefit of a coherent ecosystem more obvious).

Those are little detail that make me prefer one board rather than another.
If I have already the board, having a reminder when I download a new firmware would help too.

What do you think?

makermelissa, 2022-12-21:
If this is on a per-board basis, can't it be added to the board's description?

Olzeke51, 2023-01-20:
This would imply/dictate a 'standard' config - so that you could identify the differences.
Both Hardware and possibly software - on the software there already is a list of modules
for each board -