adafruit/circuitpython-org #1119: Allow directly programming CircuitPython onto ESP boards

makermelissa, 2022-12-23:
I talked with Scott earlier in the week and one of the things we discussed was adding the ability to program firmware directly onto ESP boards on this site. To do so, we can make use of the WebSerial ESPTool. This would make setting up devices such as the ESP32 much easier than is currently possible.

I wanted to create an issue here so I had a place to add notes. A couple of questions I've come up with so far is:
Should we install version 7 in addition to version 8?
Are we ok with just installing the .bin version for now?

The tentative answers I'm going with are that 7 could be installed, but without prompts to create a settings.toml file and .bin would be the way initially, but perhaps a bootloader with copying over the UF2 file could be implemented later.

dhalbert, 2022-12-23:
We hope 8.0.0 will be out sometime in January. Using this is the most useful for boards without USB MSC support. So I see no reason to spend time on support for 7.x.

Instead of supporting only .bin on UF2-capable boards, how about also (or instead!) adding support to erase everything and install the UF2 bootloader (i.e., do a "factory reset")? Once the bootloader is installed, then regular UF2 uploads will work.

makermelissa, 2022-12-23:
Ok, good input. By the time this is completed, 8 might be out, so the 7.x question is probably moot.

> Instead of supporting only .bin on UF2-capable boards, how about also (or instead!) adding support to erase everything and install the UF2 bootloader (i.e., do a "factory reset")? Once the bootloader is installed, then regular UF2 uploads will work.

Ok, makes sense. I'm not sure if copying the uf2 file itself is very easy at this point (it probably is doable somehow), but I'll just allow installing the bootloader automatically as an option for the initial release.

makermelissa, 2022-12-23:
I decided to take the approach of using the code from the WipperSnapper Firmware Uploader and adapting it to work on this site, which I think will be the quickest way to get it up and running. So far things are moving along nicely.

makermelissa, 2023-01-04:
So there will be a slight modification on the approach. The idea now is to change it so it's more of an install wizard like the WLED installer. Also, I'm looking at using esp-web-tools which WLED also uses for easier future maintenance and solves a couple of issues I hadn't addressed yet.

tannewt, 2023-01-04:
I was thinking we'd have an install button for each available release on the downloads page. I agree that the resulting image should include the UF2 bootloader when available.

makermelissa, 2023-01-04:
Upon further inspection, it appears that much of the wizard functionality is part of esp-web-tools, which isn't really compatible with CircuitPython because the WiFi part of the installation is handled on the device itself by the Improv WiFi library. However, I can still draw inspiration from this and go closer to my original plan.

makermelissa, 2023-02-02:
Ok, initial code upload is there. After further testing, we have fixed the CORS issues and I have updated tinyuf2 (https://github.com/adafruit/tinyuf2/pull/278, https://github.com/adafruit/tinyuf2/pull/279, and https://github.com/adafruit/tinyuf2/pull/280) so it now uploads the bootloaders to amazon S3 on new releases, which makes it available to downloads.circuitpython.org.

makermelissa, 2023-05-24:
Resolved via #149.

closed 2023-05-24 by makermelissa