adafruit/circuitpython-org #1129: Circuitpython for ESP32-C3-Devkit***

Brilipj, 2023-01-13:
The Circuitpython download page for the Espressif ESP32-C3-DevkitC-1-N4 seems to list conflicting information. https://circuitpython.org/board/espressif_esp32c3_devkitm_1_n4/
The title says ESP32-C3-DevkitC-1-N4, the test says ESP32-C3-DevkitC-02 however the URL says esp32c3_devkitm_1_n4 and the download file is named esp32_devkitm_1_n4. I have both boards but I'm not sure which one it's for. The DevkitC is based on the C3-wroom-02 on it and the DevkitM is based on the C3-mini-1.

makermelissa, 2023-01-13:
Oops, yeah. It's for the DevkitM. I probably copied the text and changed the values to be correct, but overlooked the board name itself.

Brilipj, 2023-01-13:
perfect, it went onto the C and functioned but it was much happier on the M. lol

closed 2023-01-13 by makermelissa