adafruit/circuitpython-org #1136: CircuitPython firmware download stats broken

tekktrik, 2023-01-30:
[The download stats page for the CircuitPython firmware](https://circuitpython.org/stats) appears to be broken.  Not sure what's causing the empty tables, but I'll look into it.

makermelissa, 2023-01-30:
Not sure if it's related to #1135, but the json is valid.

tekktrik, 2023-02-09:
It looks like the releases have empty lists for the `asset` key and so aren't building anything but the header: https://github.com/adafruit/circuitpython-org/blob/0400b3cc84a786baf233cd181ef2a2a9c3435018/assets/javascript/stats.js#L16-L25

dhalbert, 2023-02-09:
No, we haven't uploaded assets to GitHub for years. The download stats we use for sorting which boards are most popular  are generated by me, using AWS Athena, every time we do a release. That goes into the `download` counts in the board json.

If you want me to do other queries that update periodically I could, but it would be work to do this dynamically. Generating the download counts for all the boards takes about 15 seconds (and maybe costs about 15 cents). Maybe we should just get rid of this page. The downloads per board could be reported somewhere, if we care, since we already have that.

makermelissa, 2023-02-13:
I think updating the page to dynamically read the boards.json file via JavaScript should be possible and likely not too difficult. Automatically regenerating as static page via adabot or some similar mechanism doesn't seem practical since there would be a new PR every day that somebody would need to merge. I think the nature of this issue is as the title states the page is broken and it should just be redone if it's useful for people. I imagine something like this shouldn't take more than a couple of hours, if that.

makermelissa, 2023-02-13:
Oof, I just realized it was by CP version and not by board like I was thinking when I just wrote that. I think removing this altogether actually page makes more sense. I think as Dan stated, it costs money and time to generate and I doubt it that's useful for most people. I seems more like a fun fact.

tekktrik, 2023-02-13:
Want me to handle the PR to remove it?

PR #1147 (2023-02-15): Remove stats page
Fixes #713 
Fixes #1136
closed 2023-02-15 by makermelissa