adafruit/circuitpython-org #1163: Board feature "Castellated Pads" would be a useful filter

evildave666, 2023-03-02:
There are currently at least 40 boards with castellated pads. It would be nice to have it be a board feature in order to be able to filter to help decide what to use.

makermelissa, 2023-03-02:
Yeah, that sounds like a good feature to add. Would you like to go through the boards and submit a PR? Don't forget to update https://github.com/adafruit/circuitpython-org/blob/main/template.md so it will pass the checks. Afterwards, I can update the guide.

johnnohj, 2024-08-07:
Following up from Pull Request #1446 to echo support for including "Castellated Pads" and to propose including "Trinkey Form Factor" or similar to denote a board's capacity for direct-to-host machine connection via integrated PCB USB-A plug.

closed 2024-10-28 by dhalbert