adafruit/circuitpython-org #1170: QT PY ESP32-S3 Installer 404 file not found

bborncr, 2023-03-19:
When trying to use the online installer, the script gets stuck at `Downloading tinyuf2-adafruit_qtpy_esp32s3_nopsram-0.12.3.zip...`
Here is the debug info in the browser inspector:
```
Connecting...
base_installer.min.js:12 Connected successfully.
base_installer.min.js:12 Try hard reset.
base_installer.min.js:12 Chip type ESP32-S3
base_installer.min.js:12 Connected to ESP32-S3
base_installer.min.js:12 MAC Address: F4:12:FA:42:F8:40
base_installer.min.js:12 This chip checks out
base_installer.min.js:12 Uploading stub...
base_installer.min.js:12 Running stub...
base_installer.min.js:12 Stub is now running...
base_installer.min.js:12 Detecting Flash Size
base_installer.min.js:12 FlashId: 0x1740C8
base_installer.min.js:12 Flash Manufacturer: c8
base_installer.min.js:12 Flash Device: 4017
base_installer.min.js:12 Auto-detected Flash size: 8MB
base_installer.min.js:12 Attempting to change baud rate to 921600...
base_installer.min.js:12 Changed baud rate to 921600
cpinstaller.min.js:106 
GET https://adafruit-circuit-python.s3.amazonaws.com/bootloaders/esp32/tinyuf2-adafruit_qtpy_esp32s3_nopsram-0.12.3.zip 404 (Not Found)
```
Trying to manually download the file gives the following error in the browser:
```
https://adafruit-circuit-python.s3.amazonaws.com/bootloaders/esp32/tinyuf2-adafruit_qtpy_esp32s3_nopsram-0.12.3.zip

<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>bootloaders/esp32/tinyuf2-adafruit_qtpy_esp32s3_nopsram-0.12.3.zip</Key>
<RequestId>37A1FYFYE8H7HVXT</RequestId>
<HostId>P1GRoXYjvhq5NBLFSHcyAQcA6gRCaxzpGJ3ShmKQtOLUjyr3ewv4tUvrWYp6LINpTL7X/zGTiTc=</HostId>
</Error>
```

bborncr, 2023-03-19:
When installing another board, the process stops at the same place but in this case the error is a malformed URL. 
The URL has a double slash after `bin`.  When the second slash is removed the file downloads successfully. This installer was working flawlessly a couple of days ago.
```
Connecting...
base_installer.min.js:12 Connected successfully.
base_installer.min.js:12 Try hard reset.
base_installer.min.js:12 Chip type ESP32
base_installer.min.js:12 Connected to ESP32
base_installer.min.js:12 MAC Address: 70:B8:F6:C4:05:98
base_installer.min.js:12 This chip checks out
base_installer.min.js:12 Uploading stub...
base_installer.min.js:12 Running stub...
base_installer.min.js:12 Stub is now running...
base_installer.min.js:12 Detecting Flash Size
base_installer.min.js:12 FlashId: 0x184020
base_installer.min.js:12 Flash Manufacturer: 20
base_installer.min.js:12 Flash Device: 4018
base_installer.min.js:12 Auto-detected Flash size: 16MB
base_installer.min.js:12 ESP32 Chip only works at 115200 instead of the preferred 921600. Staying at 115200...
cpinstaller.min.js:106 
GET https://adafruit-circuit-python.s3.amazonaws.com/bin//crcibernetica-ideaboard/en_US/adafruit-circuitpython-crcibernetica-ideaboard-en_US-8.0.4.bin 404 (Not Found)
```

bborncr, 2023-03-20:
The QT Py ESP32-C3 also has problems. The only difference is that this installer never worked. The other two boards were installing just fine for several days. (BTW, all this JavaScript development for browser based esptool and repl tools is very exciting! Chromebooks are what students have to use in many countries.)
```
Failed to load resource: the server responded with a status of 404 (Not Found) stylesheet:1 
base_installer.min.js:12 Connecting...
base_installer.min.js:12 Connected successfully.
base_installer.min.js:12 Try hard reset.
base_installer.min.js:12 Chip type ESP32-C3
base_installer.min.js:12 Connected to ESP32-C3
base_installer.min.js:12 MAC Address: 34:B4:72:E7:C2:5C
base_installer.min.js:12 This chip checks out
base_installer.min.js:12 Uploading stub...
base_installer.min.js:12 Running stub...
base_installer.min.js:12 Stub is now running...
base_installer.min.js:12 Detecting Flash Size
base_installer.min.js:12 FlashId: 0x164020
base_installer.min.js:12 Flash Manufacturer: 20
base_installer.min.js:12 Flash Device: 4016
base_installer.min.js:12 Auto-detected Flash size: 4MB
base_installer.min.js:12 Attempting to change baud rate to 921600...
base_installer.min.js:12 Changed baud rate to 921600
index.js:1 Read loop got disconnected
```

closed 2023-03-20 by dhalbert

makermelissa, 2023-03-20:
> The QT Py ESP32-C3 also has problems. The only difference is that this installer never worked. The other two boards were installing just fine for several days. (BTW, all this JavaScript development [...]

I created a new issue for this in #1173