adafruit/circuitpython-org #1191: How to flash CircuitPython for SparkFun Thing Plus Matter MGM240P

lboue, 2023-05-02:
Hello,

Could you tell me how to flash the binary to the board as I was not successful.
I downloaded binary file [adafruit-circuitpython-sparkfun_thingplus_matter_mgm240p_brd2704a-en_US-8.1.0-beta.2.bin](https://downloads.circuitpython.org/bin/sparkfun_thingplus_matter_mgm240p_brd2704a/en_US/adafruit-circuitpython-sparkfun_thingplus_matter_mgm240p_brd2704a-en_US-8.1.0-beta.2.bin)

https://circuitpython.org/board/sparkfun_thing_plus_matter_mgm240p/
[image]

lboue, 2023-05-02:
I don't know if this is the right method but it seems to work :).

[image]

I just added a PR #1192 to explain how to do it.

RetiredWizard, 2023-05-02:
I'm not sure how I found my way there but I used the "In-System Programming" section on this site: https://www.silabs.com/developers/mcu-programming-options to download "Simplicity Commander" which recognized and programmed the board with the .bin file.