adafruit/circuitpython-org #1254: Simpletest Examples For Boards

FoamyGuy, 2023-09-18:
During 'in the weeds' discussion of the 8/28 meeting the idea was proposed to add board specific simpletest scripts. these would be added inside the core repo in the board directories. 

Then once they exist there, circuitpython.org website can be updated to show a button or link to the simpletest for a board on the board detail page if that board does have it's simpletest already.

The discussion can be heard in the meeting recording here: https://youtu.be/UndmYxxiNZQ?si=lldBLYo50cXk1jHX&t=1627

bato3, 2023-10-26:
At first glance, this proposal seems reasonable.

After a moment of reflection (in my situation, as a noob with circuitpython, a beginner in uC and a programmer with many years of experience), it begins to be: a lot of work that is not necessarily necessary for anyone.

Each board has a list of built-in modules that are available. Each module and library has its own very good documentation. It took me about 2 days to find documentation for the modules... And this should be the link.

FoamyGuy, 2024-10-25:
@bato3 the list of built-in modules on the downloads page now links to the documentation page for each module. 

I still believe that the device simpletest scripts would be nice as well in order to have a single place that illustrates usage of the built-in hardware on the device. The module documentation pages are great, but are split up by module, and cover only the core built-in modules so wouldn't represent any hardware that also requires a python library.  With simpletest for the device the intention is to have 1 spot to find an example of how to use all of the built-in hardware on that device rather than needing to know about which modules are used and then finding documentation for them individually.