adafruit/circuitpython-org #1263: Error: The selected drive named \ does not match the expected name of FTHRS3BOOT. Please select the correct drive.

jason-curtis, 2023-10-13:
Device: [Adafruit Feather ESP32-S3 TFT](https://circuitpython.org/board/adafruit_feather_esp32s3_tft/)

The installer halts at the UF2 installation because it doesn't recognize the drive

## Version info

Windows 11 Pro build 22621.2428 / Chrome 118.0.5993.70 

## Repro steps

* brand new Adafruit Feather ESP32-S3 TFT
* plug in to USB port with data cable
* visit Adafruit Feather ESP32-S3 TFT [CircuitPython page](https://circuitpython.org/board/adafruit_feather_esp32s3_tft/)
* open installer
* choose full install
* complete the bootloader install
* reboot the Feather. Confirm that the TFT displays a new splash screen with prompt to download the UF2 file
* click "select drive" button: [image]
* select the drive [image]
* expected behavior: it moves on to the next step an installs the UF2 file
* actual behavior: error message: `circuitpython.org says Error: The selected drive named \ does not match the expected name of FTHRS3BOOT. Please select the correct drive.`
[image]

jason-curtis, 2023-10-13:
Note: I was able to manually copy over the .UF2 and then the online tool worked fine to "update wifi credentials". That workflow requires selecting the CIRCUITPY drive and works just fine.

makermelissa, 2023-10-13:
Hmm, I wonder if this is something changed in the API recently.