adafruit/circuitpython-org #1276: Web Installer fails for Qualia (Full install)

tyeth, 2023-11-13:
### CircuitPython version

```python
9-alpha4
```

### Code/REPL

```python
``
```

### Behavior

Fails to install after warning about data loss, console error in browser tools about zip not found for tinyuf2 qualia v16

### Description

Looks for TinyUF2 for qualia, release doesn't exist.
The device (qualia) seems to show up as TFT experimenter when booted into uf2 bootloader after using factory reset.bin
Interesting to note because adafruit tft experimenter does have a tinyUF2 release.

https://github.com/adafruit/tinyuf2/issues/358

makermelissa, 2023-11-13:
Yeah, the bootloader ID (https://github.com/adafruit/circuitpython-org/blob/main/_board/adafruit_qualia_s3_rgb666.md?plain=1#L11) needs to be removed from the board file and it shouldn't try and use a file that doesn't exist.

tyeth, 2023-11-15:
I think the release exists under that bootloader id now, so do I just do a PR for the bootloader versions to bump ESP32 S3 (and maybe S2) to v0.18.1 here: https://github.com/adafruit/circuitpython-org/blob/main/_data/bootloaders.json#L13
This is the new release: 
https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-adafruit_qualia_s3_rgb666-0.18.1.zip

makermelissa, 2023-12-08:
Fixed by #1295.

closed 2023-12-08 by makermelissa