adafruit/circuitpython-org #1354: Requesting Board RSS Feed sorted by publication date

DJDevon3, 2024-03-21:
I would like to see the RSS feed in the top right corner of the CircuitPython.org website to be updated with the latest added boards. This means sorting the RSS feed in chronological order.  Most people expect an RSS feed to be in chronological order.  

The alphabetical RSS feed gives the appearance of not being updated since May 2023 if the expectation is that an RSS feed is in chronological order.  It gives the appearance of a dead RSS feed.

dhalbert, 2024-03-21:
So we would sort it by `<pubDate>`, most recent first.

makermelissa, 2024-03-21:
Yeah, that RSS feed is sorting correctly.

closed 2024-03-21 by makermelissa

DJDevon3, 2024-03-22:
Today is March 22, 2024 so this looks to be updated with the latest boards. Now I have an RSS feed for CircuitPython.org latest added boards. This is great! Thank you.

[image]

PR #1355 (2024-03-21): sort rss feed with newest at top
resolves #1354 

After this change there were two devices showing up at the top of the list (out of chronological order) It turns out the date parsing format is very specific, it needs 4 digit years and zero padded single digits, with those changes these devices now get put into the proper place of the order.

Worth considering breaking out the Blinka boards as a separate feed. Right now they effectively get rendered one after another in the same giant RSS feed. This means they won't ever appear at the top of the feed for any RSS client that doesn't do it's own [... cut]