adafruit/circuitpython-org #1376: Lolin S3 Pro - howto put this board into bootloader mode

PaulskPt, 2024-03-30:
See page [Lolin S3 PRO](https://circuitpython.org/board/lolin_s3_pro/).,
section: ```Install, Repair, or Update UF2 Bootloader```.

Note that ```Button 0``` standard is not connected to GND.
My experience is: connect ```pin 0```  (which is connected to ```Button 0```) to ```pin GND```, then perform the button presses as instructed in this section. After the board is in the bootloader mode one can undo the connection between ```pin 0``` and ```pin GND```.

FoamyGuy, 2024-10-25:
I found a schematic here: https://www.wemos.cc/en/latest/_static/files/sch_s3_pro_v1.0.0.pdf

In my understanding, it seems to indicate that the Button 0 should connect IO0 with GND when it's being pushed 
[image]

So I think it should be possible to use the instructions from the bootloader section of the downloads page:
> Press and hold down the BOOT or 0 button.
Press and release the RESET or RST button.
Release the BOOT button.

On your device if that button us not connecting GND to IO0 as expected does it connect to something different? Perhaps a trace got cut or damaged and caused it not to make successful connection. 

If this is indeed a wide spread issue that affects all of the devices we could add an extra note in the board description about using a wire to connect pin IO0 to GND instead of relying on the button. If anyone else has this device and can test please leave us a comment here letting us know if the button on your device works for entering bootloader, and if you have a multimeter to check what it connects the IO0 pin to.

PaulskPt, 2024-10-28:
Hi Tim, thank you for your investigation and time taken to create your reply. I followed the link to the schematic. Next I measured the connections of the switch with pcb marking "0" with a Ohm-meter. I can confirm that this switch on my Lolin S3 Pro board connects to GND when pushed.. I don't understand and don't remember, in this moment, how I came to the experience that moved me to create this "issue".
The board is, in this moment, flashed with: "Adafruit CircuitPython 9.1.0 on 2024-07-10; LOLIN S3 PRO 16MB Flash 8MB PSRAM with ESP32S3 Board ID:lolin_s3_pro UID:03039FD65C0F".
I made two tests to confirm the working of the "0" button. 1) situation after putting the board in "download mode" by pressing the "0" button while connecting USB cable (connected to my desktop PC). I used Espressif's flash download tool to check.  Next I brought the board back into "normal" operating mode. A CIRCUITPY D-drive popped up on my desktop PC.  See the  [imag1](https://imgur.com/a/pcKViCK).  Conclusion: it appears that creating this "issue" was not necessary. My apologies for possible too quick creating of an issue. No need to alter the text. This issue can be closed.

closed 2024-10-28 by FoamyGuy