adafruit/circuitpython-org #1404: Circuitpython Installer Documentation page

FoamyGuy, 2024-05-16:
It would be nice to have a page somewhere that documents more specifically what the choices posed by the installer mean.

[image]

I am unsure about the difference between the "Full" vs. "Upgrade" within these choices, or if there is any difference between "upgrade / install" vs. just "upgrade". 

Unless there are some rules or restrictions about specific versions I think I'd lean towards dropping the term "upgrade" altogether, to me "upgrade" implies that the process is potentially one way (i.e. increase version only), or perhaps could have rules like "only allowed to go by 1 major or minor version at a time" but as far as I know there are no restrictions on the versions, this tool could be used to go up or down to any version theoretically I think, so maybe just "Install" is a little more clear.

makermelissa, 2024-05-21:
Yeah, makes sense. Perhaps a learn guide. As it is, I'm working on some features for a package that the code editor uses, that could also be beneficial to the installer here, so there will likely be some changes to make it work on more boards. But once that's all in place, it would make sense to write something with better documentation.

FoamyGuy, 2024-10-25:
The difference between is described in a comment here: https://github.com/adafruit/circuitpython-org/issues/1440#issuecomment-2253342689

jerryneedell, 2024-10-25:
FYI - I just used the Installer for the first time on a few MagTags - These were MagTags that had not been used for quite some time and were way out of date. The Full Install worked perfectly.  Thanks for this tool.