adafruit/circuitpython-org #1440: Should "Upgrade CircuitPython Bin Only" option erase the flash?

RetiredWizard, 2024-07-26:
I've encountered this on several ESP boards, but most recently the QTPY ESP32-pico. I was expecting the difference between the "Full Circuitpython Install" and "Upgrade CircuitPython Bin Only" option to be that the Upgrade option wouldn't perform a complete flash erase first.

If that was not the intent, then it would be useful to have an option to simply re-flash the bin without the erase first. Perhaps an additional separate option to erase the flash could be added as well.

As it is, I can't really use the web flashing option in cases when I simply want to change the version of CircuitPython but don't want to loose my flash contents. I know that's a somewhat risky option but I does seem to work most of the time and is much more convenient than backing everything up and then restoring it.

Thanks!

makermelissa, 2024-07-26:
The full bin option means that it also sets up web workflow. Upgrade means it erases and flashes only. I think there are issues if the flash isn't erased, but perhaps that's not necessary.

RetiredWizard, 2024-07-26:
There are definitely times that if you don't erase the flash, it may look like it was successful but you'll have problems with the file system down the road so I understand wanting to always flash first. That being said, I do frequently risk it :grin: