adafruit/circuitpython-org #1450: "lolin S3 mini" and "lolin S3 pro" web installer links to uf2 should be updated

marcobrianza, 2024-08-11:
### CircuitPython version

```python
CircuitPython 9.1.1
```

### Code/REPL

```python
not related to code
```

### Behavior

web installer does not work

### Description

web installer does not complete since it tries to download a file that does not exist. the correct .uf2 now exists since 2 days but the web install script should be updated accordingly.

dhalbert, 2024-08-11:
This is in process after https://github.com/adafruit/tinyuf2/issues/401. There are some files to test in https://github.com/adafruit/tinyuf2/releases/tag/0.19.0. @marcobrianza Could you test those and see if they work for you.

marcobrianza, 2024-08-11:
Hi, consider I'm just starting to use circuitpython that is why I was counting very much on the web installer.
anyway I flashed the combined.bin using https://adafruit.github.io/Adafruit_WebSerial_ESPTool/
then I copied to the pseudo-usb-drive adafruit-circuitpython-lolin_s3_mini-en_GB-9.1.1.uf2

circuitpython runs and I just made the onboard neopixel blink.

board.out is the follwing:

Adafruit CircuitPython 9.1.1 on 2024-07-23; LOLIN S3 MINI 4MB Flash 2MB PSRAM with ESP32S3
Board ID:lolin_s3_mini
UID:CD45576D514D

the board is recognised by the system as following
[image]

marcobrianza, 2024-08-11:
I can also test "lolin s3 PRO" but there is no new release for it

tannewt, 2024-09-05:
This will be fixed by https://github.com/adafruit/circuitpython-org/pull/1469

marcobrianza, 2024-09-06:
HI, I see the links are updated but there is no https://github.com/adafruit/tinyuf2/releases/download/0.20.1/tinyuf2-lolin_s3_pro-0.20.1.zip file (the s3 mini works correctly)

makermelissa, 2024-09-06:
I don't see a lolin_s3_pro under the release assets at https://github.com/adafruit/tinyuf2/releases. Either the bootloader on that board is incorrect or it hasn't been added to tinyuf2. Is it possible it just uses the `lolin_s3` bootloader?

marcobrianza, 2024-09-06:
I have flashed a "lolin s3 pro" from the "lolin s3" download page with the web installer. I got circuitpython working this way.
I am not sure what does the bootloader need to access, anyway the RGB LED is on gpio 38 in both boards.

makermelissa, 2024-09-06:
> I have flashed a "lolin s3 pro" from the "lolin s3" download page with the web installer. I got circuitpython working this way. I am not sure what does the bootloader need to access, anyway the RGB LED is on gpio 38 in both boards.

Ok, we should just point to that bootloader for now then. I'll create a PR to fix it.

PR #1470 (2024-09-06): Fix lolin s3 pro bootloader
Fixes #1450. There is no lolin_s3_pro, but @marcobrianza has successfully used the lolin_s3 bootloader.
closed 2024-09-06 by dhalbert