adafruit/circuitpython-org #1572: Allow to chose the binary in iweb installer.

0wwafa, 2025-01-17:
Web installer is nice but only updates to a fixed (latest) binary.
Please add a button to select a local binary.

Neradoc, 2025-01-17:
Hi, I believe you are talking about the installer on circuitpython.org
Here is a web version of esptool to install binaries, and a guide:
https://adafruit.github.io/Adafruit_WebSerial_ESPTool/
https://learn.adafruit.com/circuitpython-with-esp32-quick-start/web-serial-esptool

I'm not sure where it's linked to on the site. It is referenced when there's a UF2 bootloader section. Maybe we could link to it for ESP32 boards without a UF2 bootloader in particular, since it's the main way to install the "Absolute Newest" and "Previous Versions" (outside of using esptool.py).

tannewt, 2025-01-17:
CircuitPython.org is meant for CircuitPython. We don't want to have to support other random binaries.

0wwafa, 2025-01-18:
> CircuitPython.org is meant for CircuitPython. We don't want to have to support other random binaries.

I didn't mean random binaries, but for example the latest or previous versions.

makermelissa, 2025-12-18:
Looking into this, it may be possible. We can get a list of releases from https://api.github.com/repos/adafruit/circuitpython/releases. My concern is that the GitHub API limit may be too limiting, but it appears to be 60 per hour per IP Address for unauthenticated (i.e. no token) access, which shouldn't affect most users. If need be, in the future, we could implement OAuth over Javascript, but it complicates the code and likely isn't necessary.

Anyway, once we have the release info, we should be able to put together the correct URL for accessing it with S3.

dhalbert, 2025-12-18:
We could enumerate the releases for any particular board by looking at https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin and the board dir, with an S3 API. Or we could add some Javscript there to enumerate them. The file lister right now is already some server-side JavaScript. So I think we could avoid going to GitHub.

makermelissa, 2025-12-18:
Ok, I do like that solution better.