adafruit/circuitpython-org #1618: Homepage doesn't display CircuitPython logo on Firefox

todbot, 2025-04-11:
This is what I see on Firefox.  "CircuitPython Logo" text instead of https://circuitpython.org/assets/images/logo-dark.png
[image]

dhalbert, 2025-04-11:
Can you do a Shift-reload? I see it on Firefox 137.0 on Linux and also on Firefox 137.0.1 on macOS 15.4.

todbot, 2025-04-11:
Similar results in Safari iOS. Safari MacOS looks as expected.

dhalbert, 2025-04-11:
I'm still not seeing this problem, either on iOS (Safari or Firefox), or any other platform. Do you have an iOS phone? Any difference between on WiFi and cellular data?

@jwcooper Any ideas about this?

Neradoc, 2025-04-11:
Can't repro on Firefox or Safari, does it load on reloading the page ?
Can you have the "network" panel open in firefox when loading the page and not seeing the logo ?
Any errors in the console ?

I can actually see an error in the srcset:
```
/assets/images/logo-dark.png 1x, /assets/images/logo-dark@2x.png 2x, /assets/images/logo-dark3x.png 3x
```
The 3rd one should be `logo-dark@3x.png`. 
If I understand the srcset, it picks an image depending on the screen "retina" type. On my mac I have 4k display in 1080p so it picks the `logo-dark@2x.png`, which works. But on higher density like a recent iPhone (mine is a XS) it might take the 3x, which is 404.

dhalbert, 2025-04-11:
@todbot should be fixed, if you can test with a shift-reload. Your displays are too good!

Thanks @Neradoc!!

closed 2025-04-11 by dhalbert

todbot, 2025-04-11:
Yep fixed! Awesome!

PR #1619 (2025-04-11): Fix logo srcset for 3x
- Fixes #1618

Thanks @Neradoc for sleuthing