adafruit/circuitpython-org #1634: No Blinka Boards appear to be added for several months

TheKitty, 2025-05-13:
For the Python on Microcontrollers Newsletter I go to the Blinka section then "Sort by Newest First.

For months the board here is the newest, indicating no boards have been added https://circuitpython.org/blinka/waveshare_rp2040_one/

Is it true that no boards have been added in months, or is the sort function broken, or perhaps Blinka boards have not been submitted?

Thanks.

makermelissa, 2025-05-13:
There are some new ones I merged in recently, however I just need to add them manually. I will likely work on that as soon as I'm done with the guide I'm finishing up.

makermelissa, 2025-05-13:
I'll also verify the sorting is working at the same time because the latest board has a date of July last year.

makermelissa, 2025-05-13:
Ok, looks correct. The last PR Blinka boards were added was in https://github.com/adafruit/circuitpython-org/pull/1436.

makermelissa, 2025-05-16:
Fixed via #1636